multigcs/riocore #5: esp32 bridge alternative to w5500

jschoch, 2024-02-28:
perhaps UDP over USB is viable for bridge connection to LCNC host?

https://github.com/wezhunter/ESP32_LinuxCNC_MotionController_RealTime/issues/14

wezhunter, 2024-02-29:
Yes it is and it will be! Nice spot! Even if its not used as a motion controller I reckon it will be a nice powerful low latency wireless accessory tool for LinuxCNC 😄

wezhunter, 2024-02-29:
Sorry commented on the wrong repo. Darn github reference links. My bad. Must get sleep.

multigcs, 2024-02-29:
@wezhunter 
haahaa, i was just confused too, i got an email from github with a text from your repo but with a link to mine 

```
Thanks for the PR.
Sorry for late reply. I've been hectic working on a big commit with lots of changes and bug fixes - including USB S2/S3 support.
Once that's complete it will be much smaller increments/fixes so contribution is easy. I'll definitely take a look at your PR soon

```

multigcs, 2024-02-29:
@jschoch 
sounds good !

jschoch, 2024-03-18:
closing this, it seems easier to just connect the esp32 directly to LCNC unless you need RT.

closed 2024-03-18 by jschoch