multigcs/riocore #8: sticky config or config history or config metadata

jschoch, 2024-03-18:
finding it kind of hard to figure out what config went with something I did a few weeks ago.  I have several machines (3) and multiple places I've run riocore/riogui and it is easy to forget what has been done.

Ideas:

copy config to output dir, put it in a .config dir if you  want to hide it.  maybe even keep older versions around.
copy config filename and or file path and or checksum somewhere in Output.
embed RIO gui commit/release or use semantic versioning tag so you have some idea what version of rio* was used for config and or Output dir.

multigcs, 2024-03-18:
# cat Output/Tangoboard/.config.json   ;)

jschoch, 2024-03-18:
documenting that would have been nice ;)

ok, feel free to close if you don't see the value in capturing the source path.

multigcs, 2024-03-18:
no no, it's a good idea to have a history and diffs !

digiexchris, 2024-10-31:
I store my config.json (and modified linuxcnc ini) in a private git repo, built in history and diffs :)