multigcs/riocore #10: Startup CLK for spartan6

ozzyrob, 2024-07-02:


multigcs, 2024-07-03:
thanks, have adding new option to the boardconfig:
```
    "clock": {
        "speed": "50000000",
        "pin": "p55",
        "startup": "CClk"
    },
```

if you send me your .ucf file,
maybe i can add the missing stuff to the generator 'riocore/generator/pins/ucf/pins.py'

ozzyrob, 2024-07-03:
Dude you are awesome !
Thank you so much
Something may seem a little different in this file as I'm using a different Spartan 6 board
[pins.ucf.txt](https://github.com/user-attachments/files/16080623/pins.ucf.txt)

New Board for Mimas 
[Mimas Spartan 6 Product Page](https://numato.com/product/mimas-spartan-6-fpga-development-board/)
[Mimas.json.txt](https://github.com/user-attachments/files/16080644/Mimas.json.txt)

Basic config for the board
[config-test.json.txt](https://github.com/user-attachments/files/16080658/config-test.json.txt)

multigcs, 2024-07-03:
I have to thank you :)

I have added an option for the pins,
at least for ISE for now, 
to be able to set my own values for "screw", "drive" and "iostandard":
Example:
```
        {
            "type": "stepdir",
            "name": "Joint0",
            "pins": {
                "step": {
                    "pin": "BOB_A:P2",
                    "screw": "FAST",
                    "drive": 24,
                    "iostandard" = "LVTTL"
                },
                "dir": {
                    "pin": "BOB_A:P3"
                    "drive": 24,
                    "iostandard" = "LVTTL"
                }
            },
            "is_joint": true
        },
```

now it is also possible now to set this values in the plugin.py files
and i will try to set some good defaults there:

```
        self.NAME = "blink"
        self.VERILOGS = ["blink.v"]
        self.PINDEFAULTS = {
            "led": {
                "direction": "output",
                "drive": 8,
            },
        }
```

defaults for outputs are : 

  IOSTANDARD = LVTTL | DRIVE = 4 | SLEW = SLOW

for inputs:

  IOSTANDARD = LVTTL

ozzyrob, 2024-07-03:
Once again that's great work, and fast.
Should "screw" be "slew" ?

There is also a PULLUP or PULLDOWN attribute, would it be worth including it for the Spartan 6
[Spartan-6 FPGA SelectIO Attributes/Constraints](https://opencores.org/websvn/filedetails?repname=openmsp430&path=%2Fopenmsp430%2Ftrunk%2Ffpga%2Fxilinx_avnet_lx9microbard%2Fdoc%2FSpartan-6_SelectIO_Resources_ug381.pdf#G4.278000)

multigcs, 2024-07-03:
sorry,  'slew' :)

pullup/pulldown support for ucf is added

multigcs, 2024-07-03:
pullup/pulldown are two boolean options now, i think i rewrite this part:

from:

pullup: true
pulldown: false

-> 

pull: "UP"
pull: "DOWN"
pull: "NONE"

ozzyrob, 2024-07-03:
Perfect, man you work fast.

Just a little off topic if that's ok, with UDP what's the better option WT32_ETH0 bridge solution or a Wiznet W5500 module ?
I have both modules available for use.

multigcs, 2024-07-03:
at the moment i think WT32_ETH0,
it's little bit slower but rock solid (last test: zero packet loss in 2days)

but the W5500 is well suited for testing, no real problems, no connection interruptions

ozzyrob, 2024-07-03:
Okay I'll give it a go.
Thanks

ozzyrob, 2024-07-05:
Gave the W5500 ago, ping times about half of WT32_ETH01, but connection not as stable, once again SPI cables were not the best, long and floaty. Once I get my preferred dev board back I'll give it a proper try.

closed 2024-07-25 by multigcs