multigcs/riocore #12: w5500

ozzyrob, 2024-07-11:
I was looking at the timing diagrams for the W5500 and was wondering if it would be worth inserting a small delay from when cs is taken low and the SPI clock starts. I know the data sheets suggest a min of about 7ns or so. 
Obviously not having this is not an issue when configuring the device, as it does this fine, but could this be causing the issue with the driver reporting "wrong header" once Linuxcnx is up and running ?
I've tried various pull-up resistors and changing the divider to lower the SPI clock speed but nothing appears to change. Unfortunately I don't have the required equipment to monitor the signals on the SPI bus.
Just to add I'm not using long Dupont cables, I'm using a PCB with short wire wrap jumpers. I have used this method before to get reliable comms in the high teens of MHz connecting a RPi4 to a diy 7c81.

The more I looked at FPGA code the more I getting to understand it.

multigcs, 2024-07-11:
good idea, i'll give it a try, thank you very much.

I also have a board with very short connections and still the problems, with the esp I also drive higher rates via SPI without problems, it really is a strange problem

ozzyrob, 2024-07-11:
Cheers, this the timing diagram, in case you haven't seen it yet.
https://cdn.sparkfun.com/datasheets/Dev/Arduino/Shields/W5500_datasheet_v1.0.2_1.pdf#%5B%7B%22num%22%3A126%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C82%2C751%2C0%5D

ozzyrob, 2024-07-16:
Sorry to bang on about this:

I was looking at this code, my naive understanding is that miso should be put into an index in rxbuffer & data_read, but to me it would seem it's being read into the same "bit position", I would of thought that spi_count should be accounted for in the index, but what seems strange is that it would seem to work.
```
always @(posedge clk) begin
        if (spi_clk == 1'b0 && state == STATE_SENDING_COMMAND && spi_clock_count >= 24 && spi_clock_count <= 31) begin
            data_read <= {data_read[6:0], miso};
        end else if (spi_clk == 1'b0 && state == STATE_PULLING_DATA && spi_clock_count) begin
            rx_buffer <= {rx_buffer[BUFFER_SIZE_RX+HEADER_SIZE+24 - 2:0], miso};
        end
    end
```
I thought it would be something along the lines of:
rx_buffer[_bitcount_] <= miso;
_bitcount_ - 1b'1;
_Then we check that bitcount is 0_

multigcs, 2024-07-16:
I think it's good when someone else looks at it !

a line like this:
data_read <= {data_read[6:0], miso};

represents a shift register

8bit <= {7bit, 1bit}

closed 2024-07-25 by multigcs