multigcs/riocore #16: $clog2 Not supported by xst - Xilinx tools

ozzyrob, 2024-08-03:
Just trying to use the dev branch for testing the wiznet code, apparently the xilinx tools only seem to support Verilog-2001.

https://support.xilinx.com/s/article/44586?language=en_US

multigcs, 2024-08-03:
thanks for the report, as a workaround to test, you can replace the parts:

$clog2(...) with 31

i will check if i find a solution, if not, i revert this changes

multigcs, 2024-08-03:
or go back one commit:
git checkout 9fe64ce114abb9471fb35c795719ba19b66bd558

multigcs, 2024-08-03:
Fixed :) 
i hope it works for all toolchains

https://github.com/multigcs/riocore/commit/ecf2f3a7dfb758535c51ab60127b24876c019067

multigcs, 2024-08-03:
seems to work with all except altera/quartus :(

closed 2024-08-03 by multigcs

ozzyrob, 2024-08-04:
Build went ok, now I can test hardware :)