multigcs/riocore #24: Compatibility Inquiry: ALTERA Cyclone II FPGA with LinuxCNC-RIO

Tpj-root, 2024-09-25:
Sir,
I have an ALTERA FPGA Cyclone II EP2C5T144 board. Is it possible to use this with LinuxCNC-RIO for real-time I/O? If yes, could you provide any guidance or steps to make it work?

Thanks!

multigcs, 2024-09-25:
not shure, 
the problem is that ‘Cyclone II’ is no longer supported by quartus and i don't know if rio works with the old version.

multigcs, 2024-09-25:
in this zip-file i have a test gateware with makefile,
if this runs with the old quartus than we can add support for this fpga:
http://multixmedia.org/EP2C5T144.zip

multigcs, 2024-09-25:
ok, i installed this version: 
https://www.intel.com/content/www/us/en/software-kit/711790/intel-quartus-ii-web-edition-design-software-version-13-0sp1-for-linux.html

and by removing some lines from pins.qdf (..IO_STANDARD..) i can build the bitfile.

so it looks good, but i can not test it.

multigcs, 2024-09-25:
Ok, i have add support for this board :)

https://github.com/multigcs/riocore/commit/afc4bd855d440148068b1e432f1e019f482381ae
https://github.com/multigcs/riocore/commit/2849eedc1c1d7464477d58fa2bb9e2861c62d59e

but please tell me if it realy works, the demo config is only for blinking the LED

riocore/configs/EP2C5T144/config.json

Tpj-root, 2024-09-26:
Sir, thank you for the immediate response. Please give me a few days. The board is available in the hackerspace lab, and I have requested them to send it to me. After I receive the package, I will try to install it and report to you on how it works..

Tpj-root, 2024-09-29:
[image]
[image]

Sir, I finally received my package today and successfully compiled it, but I encountered an error. How can I fix this?

multigcs, 2024-09-30:
hi, the flash command depends a lot on your hardware, you have to see how it is normally loaded on the fpga, maybe try OpenFPGAloader

Tpj-root, 2024-10-01:
Sir,

I uploaded the .pof file using my friend's Windows machine, and it was uploaded successfully.

[image]

After that, I tested it with an oscilloscope:

[image]

- PIN 17: CH1 (clock)
- Push button on PIN 144: CH2 (I get a signal when pressed)

I think it’s working. Now, I am waiting for your response to confirm if it’s really working.

multigcs, 2024-10-01:
is one of the LED's blinking (1Hz) ? then it's working 

next step is setup something for the host connection (SPI / W5500), and create some IO's like StepDir or PWM outputs.

to flash, you can also try 
```
cd Output/EP2C5T144/Gateware
make prog
```
this will use the 'quartus_pgm' command to load the bitstream

Tpj-root, 2024-10-01:
sir, To test the LED blinking using a push button and a 1kΩ pull-up resistor.
[image]

multigcs, 2024-10-01:
the LED have to blink constantly with 1Hz after flashing, no SW / PullUp / ... needed.

multigcs, 2024-10-01:
but not shure my test bitfile have the blinking LED part, you have to build and compile your own bitfile

riocore/configs/EP2C5T144/config.json

multigcs, 2024-10-01:
after 'usermod' you have to logout and re-login your user.

what programmer do you have ?

Tpj-root, 2024-10-01:
sir, I have an Altera USB Blaster.
[image]

multigcs, 2024-10-01:
hmm, ok, normaly it should work

Tpj-root, 2024-10-03:
Sir, after using OpenFPGALoader to manually upload the .svf file, the changes are temporary. Now the LED is blinking. Could you please confirm if it worked?

[image]
[image]

https://github.com/user-attachments/assets/5092d33b-e96f-4392-b03f-500fab72f58e

multigcs, 2024-10-03:
looks very good :)
try to add '-f' option to your openFPGAloader command to write to flash

Tpj-root, 2024-10-04:
Sir, The -f command does not work.
I think .sof and .svf files are not used for permanently programming the FPGA.
I need to use a file format that writes to non-volatile memory, such as .pof

.sof vs .pof
Both files show significant differences in their hex structure.

I believe the only way to permanently program the FPGA is by using the .pof file.
It seems to be the only solution.

Sir, I need your guidance.

I want to continue with this setup: LinuxCNC_RIO with the .svf file. After learning RIO and all the fundamentals (config, modules, and plugins), I will jump into RIOctrl FPGAs.

Can I continue? Or which path is perfect for learning the fundamentals?
Sir, Thank you for spending your precious time to answer my questions.

multigcs, 2024-10-04:
what do you mean by RIOctrl FPGAs ?

Tpj-root, 2024-10-04:
Sir, after that, I became familiar with RioCore. Is RIOctrl FPGAs suitable for advanced-level work in making CNC machines?

multigcs, 2024-10-04:
you mean this board !?!?
https://github.com/festlv/rioctrl-hw

as far as i know, the basic parts are already working, but it's still in the test phase

closed 2024-12-31 by multigcs