multigcs/riocore #42: w5500 issue on tang 9k

jschoch, 2024-12-21:
I have a working setup using commit:

I'm trying to get a new setup going but when I flashed the network is mostly not working, though I have seen a few packets make it.

The new env has the current repo's code in it, it also has 

Gowin_V1.9.10.03_Education_linux.tar.gz  vs Gowin_V1.9.9Beta-4_Education.tar.gz 
and 

oss-cad-suite-linux-x64-20241220.tgz   vs oss-cad-suite-linux-x64-20240229.tgz

I tried slowing down the clock but that doesn't seem to work

since it just throws network errors i'm not quite sure how to troubleshoot.  I have something working so no rush on this.

Full log here

```
rm -rf rio.fs rio.json rio_pnr.json rio.tcl abc.history impl
gw_sh rio.tcl
*** GOWIN Tcl Command Line Console  ***
current device: GW1NR-9C  GW1NR-LV9QN88PC6/I5
add new file: "globals.v"
add new file: "quadencoderz.v"
add new file: "blink.v"
add new file: "stepdir.v"
add new file: "w5500.v"
add new file: "debouncer.v"
add new file: "toggle.v"
add new file: "pwmmod.v"
add new file: "rio.v"
add new file: "rio.sdc"
add new file: "pins.cst"
GowinSynthesis start
Running parser ...
Analyzing Verilog file '/home/schoch/dev/riocore/Output/lathe_quadz_expand/Gatew                                                are/globals.v'
WARN  (EX3209) : Root scope declaration is not allowed in Verilog 2001 mode("/ho                                                me/schoch/dev/riocore/Output/lathe_quadz_expand/Gateware/globals.v":1)
WARN  (EX3209) : Root scope declaration is not allowed in Verilog 2001 mode("/ho                                                me/schoch/dev/riocore/Output/lathe_quadz_expand/Gateware/globals.v":11)
Analyzing Verilog file '/home/schoch/dev/riocore/Output/lathe_quadz_expand/Gatew                                                are/quadencoderz.v'
Analyzing Verilog file '/home/schoch/dev/riocore/Output/lathe_quadz_expand/Gatew                                                are/blink.v'
Analyzing Verilog file '/home/schoch/dev/riocore/Output/lathe_quadz_expand/Gatew                                                are/stepdir.v'
Analyzing Verilog file '/home/schoch/dev/riocore/Output/lathe_quadz_expand/Gatew                                                are/w5500.v'
Analyzing Verilog file '/home/schoch/dev/riocore/Output/lathe_quadz_expand/Gatew                                                are/debouncer.v'
Analyzing Verilog file '/home/schoch/dev/riocore/Output/lathe_quadz_expand/Gatew                                                are/toggle.v'
Analyzing Verilog file '/home/schoch/dev/riocore/Output/lathe_quadz_expand/Gatew                                                are/pwmmod.v'
Analyzing Verilog file '/home/schoch/dev/riocore/Output/lathe_quadz_expand/Gatew                                                are/rio.v'
Compiling module 'rio'("/home/schoch/dev/riocore/Output/lathe_quadz_expand/Gatew                                                are/rio.v":40)
WARN  (EX3780) : Using initial value of 'ESTOP' since it is never assigned("/hom                                                e/schoch/dev/riocore/Output/lathe_quadz_expand/Gateware/rio.v":71)
WARN  (EX3791) : Expression size 24 truncated to fit in target size 23("/home/sc                                                hoch/dev/riocore/Output/lathe_quadz_expand/Gateware/rio.v":92)
Compiling module 'quadencoderz(QUAD_TYPE=2)'("/home/schoch/dev/riocore/Output/la                                                the_quadz_expand/Gateware/quadencoderz.v":2)
Compiling module 'blink(DIVIDER=13500000)'("/home/schoch/dev/riocore/Output/lath                                                e_quadz_expand/Gateware/blink.v":2)
WARN  (EX3791) : Expression size 32 truncated to fit in target size 25("/home/sc                                                hoch/dev/riocore/Output/lathe_quadz_expand/Gateware/blink.v":17)
Compiling module 'stepdir'("/home/schoch/dev/riocore/Output/lathe_quadz_expand/G                                                ateware/stepdir.v":2)
WARN  (EX3073) : Port 'data_read' remains unconnected for this instance("/home/s                                                choch/dev/riocore/Output/lathe_quadz_expand/Gateware/w5500.v":99)
Compiling module 'w5500(BUFFER_SIZE=16'b0000000010010000,IP_ADDR=32'b11000000101                                                010000000101011000010,NET_MASK=32'b11111111111111111111111100000000,GW_ADDR=32'b                                                11000000101010000000101000000001,MAC_ADDR=48'b1010101010101111111110101100110011                                                10001100011100,DIVIDER=10)'("/home/schoch/dev/riocore/Output/lathe_quadz_expand/                                                Gateware/w5500.v":3)
WARN  (EX3791) : Expression size 32 truncated to fit in target size 5("/home/sch                                                och/dev/riocore/Output/lathe_quadz_expand/Gateware/w5500.v":35)
Compiling module 'wiznet5500(IP_ADDR=32'b11000000101010000000101011000010,NET_MA                                                SK=32'b11111111111111111111111100000000,GW_ADDR=32'b1100000010101000000010100000                                                0001,MAC_ADDR=48'b101010101010111111111010110011001110001100011100,BUFFER_SIZE_R                                                X=16'b0000000010010000,BUFFER_SIZE_TX=16'b0000000010010000,MSGID=32'b01110100011                                                010010111001001110111)'("/home/schoch/dev/riocore/Output/lathe_quadz_expand/Gate                                                ware/w5500.v":105)
WARN  (EX3780) : Using initial value of 'local_ip' since it is never assigned("/                                                home/schoch/dev/riocore/Output/lathe_quadz_expand/Gateware/w5500.v":256)
WARN  (EX3780) : Using initial value of 'local_gw' since it is never assigned("/                                                home/schoch/dev/riocore/Output/lathe_quadz_expand/Gateware/w5500.v":257)
WARN  (EX3780) : Using initial value of 'net_mask' since it is never assigned("/                                                home/schoch/dev/riocore/Output/lathe_quadz_expand/Gateware/w5500.v":258)
WARN  (EX1998) : Net 'tx_buffer_write_pointer[0]' does not have a driver("/home/                                                schoch/dev/riocore/Output/lathe_quadz_expand/Gateware/w5500.v":265)
NOTE  (EX0101) : Current top module is "rio"
WARN  (EX0211) : The output port "rst" of module "w5500(BUFFER_SIZE=16'b00000000                                                10010000,IP_ADDR=32'b11000000101010000000101011000010,NET_MASK=32'b1111111111111                                                1111111111100000000,GW_ADDR=32'b11000000101010000000101000000001,MAC_ADDR=48'b10                                                1010101010111111111010110011001110001100011100,DIVIDER=10)" has no driver, assig                                                ning undriven bits to Z, simulation mismatch possible("/home/schoch/dev/riocore/                                                Output/lathe_quadz_expand/Gateware/w5500.v":21)
[5%] Running netlist conversion ...
Running device independent optimization ...
[10%] Optimizing Phase 0 completed
[15%] Optimizing Phase 1 completed
[25%] Optimizing Phase 2 completed
Running inference ...
[30%] Inferring Phase 0 completed
[... 699 lines cut ...]
                "bit": {
                    "display": {
                        "title": "BOB DIN1"
                    },
                    "net": "bob_din1_P10"
                }
            },
            "name": "BOB_DIN1_P11"
        },
        {
            "type": "bitin",
            "pins": {
                "bit": {
                    "pin": "33",
                    "pullup": true
                }
            },
            "name": "BOB_DIN0_P10"
        }
    ],
    "machinetype": "lathe",
    "axis": 2,
    "joint_n": 0
}
```

jschoch, 2024-12-22:
Tried the 1.9.9 gowin education toolchain, ever so slightly better but still not usable.  Trying the old oss-cad-suite next.

jschoch, 2024-12-22:
downgrading cadsuite didn't help, trying to get old commit hash to work on new machine next.

jschoch, 2024-12-22:
downgrading to short hash "4713f92"  commit from the old riocore with the old riogui repo that I had to copy over since it is gone has the w5500 driver working great on the new machine....    

Not sure what to try next, maybe re-upgrade the GOWIN and oss-cad-suite packages and see if the old commit still works, then try some later commits to see where it breaks?

jschoch, 2024-12-22:
097491a is working... trying something later.

jschoch, 2024-12-22:
9fe64ce114abb9471fb35c795719ba19b66bd558    seems broken, going earlier....

jschoch, 2024-12-22:
e0bae738eddf2fe2e7979a4949cd0d5f5d10829b  borked
c3889511f2841413ee1d767b3223f9d140048087  couldn't get it to generate

I tried to copy the w5500 dir from plugins from commit 097491a with the current HEAD but I still get network issues.  Tried this with the updated oss-cad-suite also.   

the error that is jumping out to look at is `WARN  (TA1132) :  'w550014/mclk' was determined to be a clock but was not create` because I don't see that in the logs of a working build.

If I flash with full checkout of 097491a  I can run the test-gui with HEAD, so that kinda rules out an issue there.

more digging....

jschoch, 2024-12-22:
I think c388951 works, but I have to generate with 097491a.  This leads me to believe that between c388951 (July 20) and e0bae73 the breaking change happened.   Very odd that I can't just use the old plugin in HEAD but I think we have a fairly narrow commit window for the issue.

jschoch, 2024-12-22:
314edbe is working, i was going to modify it to match c388951, but I forgot to and was compiling when i realized but said f-it and just flashed it, but amzingly it worked!  oh boy... confusing.

multigcs, 2024-12-22:
wow, thanks for the effort you have made, unfortunately I can't quite understand it today, I'll look at it tomorrow or the next few days

jschoch, 2024-12-31:
the i2c_bus branch is working with my w5500 so i'm closing this.

multigcs, 2024-12-31:
i need this as a reminder, will check the timing of mclk

closed 2025-04-09 by multigcs