multigcs/riocore #44: QUAD_TYPE borked

jschoch, 2024-12-31:
So, to sniff out the bug a bit further I added a pin to show the delta counts for every run of the rps calculation.  It showed that delta counts was always 32 no matter what the actual speed was of the pulses.  This didn't match up with the UDP latency and would basically only calculate the velocity every (pulse_duration * 32), this means that for slow updates the data was sparser and sparser thus requiring filtering.  I did a bunch of digging and couldn't find anything obvious so as a last resort (and not to bug you olli) i asked gemini.  Amazingly it was correct.  Removing the QUAD_TYPE bitshift gave the expected updates for every udp packet!

simple change
```verilog
//assign position = $signed(count>>>QUAD_TYPE);
assign position = count;
```

I asked gemini "why would <quadencoderz.v> only send counts when the counts diff was 32?"

it said:

<html><body>
<div id=":n5" class="Am aiL Al editable LW-avf tS-tW tS-tY" hidefocus="true" aria-label="Message Body" writingsuggestions="false" g_editable="true" role="textbox" aria-multiline="true" contenteditable="true" style="direction: ltr; min-height: 1103px;" tabindex="1" spellcheck="false" aria-owns=":s3" aria-controls=":s3" aria-expanded="false"><p>The issue lies in the line:</p><u></u><u></u><div style="display:block"><div><span>Verilog</span><div><button aria-label="Copy code"><span></span><u></u><u></u><span></span><span></span></button></div></div><div><div><pre><code role="text"><span>assign</span> position = <span>$signed</span>(count&gt;&gt;&gt;QUAD_TYPE);
</code></pre></div></div></div><u></u><u></u><p>You're using a right arithmetic shift (<code>&gt;&gt;&gt;</code>) with the parameter <code>QUAD_TYPE</code>. This is intended to divide the count by 2 raised to the power of <code>QUAD_TYPE</code>.</p><p>If <code>QUAD_TYPE</code> is set to 5 (2^5 = 32), then the <code>position</code> output will only change when the <code>count</code> changes by 32. Let's break down why:</p><ul>
<li><b><code>count&gt;&gt;&gt;QUAD_TYPE</code>:</b> This performs an arithmetic right shift. It shifts the bits of <code>count</code> to the right by <code>QUAD_TYPE</code> positions. For each shift, the leftmost bit (the sign bit) is replicated, preserving the sign of the number.</li>
<li><b>Effect of the Shift:</b> When <code>QUAD_TYPE</code> is 5, the shift effectively divides <code>count</code> by 32. So:
<ul>
<li>If <code>count</code> is 0, <code>position</code> is 0.</li>
<li>If <code>count</code> is 1, <code>position</code> is 0 (1 &gt;&gt;&gt; 5 = 0).</li>
<li>...</li>
<li>If <code>count</code> is 31, <code>position</code> is 0 (31 &gt;&gt;&gt; 5 = 0).</li>
<li>If <code>count</code> is 32, <code>position</code> is 1 (32 &gt;&gt;&gt; 5 = 1).</li>
<li>If <code>count</code> is 63, <code>position</code> is 1 (63 &gt;&gt;&gt; 5 = 1).</li>
<li>If <code>count</code> is 64, <code>position</code> is 2 (64 &gt;&gt;&gt; 5 = 2).</li>
</ul>
</li>
</ul><p>As you can see, <code>position</code> only increments when <code>count</code> has increased by a multiple of 32. The same applies to decrements.</p><p><b>How to Fix It:</b></p><p>If you want <code>position</code> to reflect the raw encoder count, you should remove the shift entirely:</p><u></u><u></u><div style="display:block"><div><span>Verilog</span><div><button aria-label="Copy code"><span></span><u></u><u></u><span></span><span></span></button></div></div><div><div><pre><code role="text"><span>assign</span> position = count; <span>// Assign the count directly</span>
</code></pre></div></div></div><u></u><u></u><p>If you intend to divide the count by a fixed value (e.g., to implement quadrature decoding where you get 4 counts per encoder cycle), you should use a division or a right shift by a <i>constant</i> value, not a parameter that can be easily misinterpreted:</p><u></u><u></u><div style="display:block"><div><span>Verilog</span><div><button aria-label="Copy code"><span></span><u></u><u></u><span></span><span></span><span></span></button></div></div><div><div><pre><code role="text"><span>// For quadrature decoding (4 counts per cycle):</span>
<span>assign</span> position = count &gt;&gt;&gt; <span>2</span>; <span>// Right shift by 2 (equivalent to division by 4)</span>

<span>// Or, for division by a specific value (e.g., 10):</span>
<span>assign</span> position = count / <span>10</span>; <span>// Use division (more resource-intensive in hardware)</span>
</code></pre></div></div></div><u></u><u></u><p><b>Example demonstrating the shift:</b></p><p>Let's say <code>count</code> is a 6-bit number and <code>QUAD_TYPE</code> is 2:</p><div>
<div><u></u><u></u><div><div>
count (Binary) | count (Decimal) | count >>> 2 (Binary) | count >>> 2 (Decimal)
-- | -- | -- | --
000000 | 0 | 000000 | 0
000001 | 1 | 000000 | 0
000010 | 2 | 000000 | 0
000011 | 3 | 000000 | 0
000100 | 4 | 000001 | 1
000101 | 5 | 000001 | 1
... | ... | ... | ...

</div></div><u></u><u></u></div>
</div><p>This table clearly shows how the right shift with <code>QUAD_TYPE = 2</code> effectively divides the count by 4.</p><p>By making the appropriate change (either removing the shift or using a constant value), you'll get the desired behavior from your quad encoder module. If you intended to divide by 32 based on some external configuration, consider using a separate signal and a conditional assignment instead of relying on a parameter that affects the core counting logic.</p></div>
</body>
</html>

multigcs, 2025-01-01:
default QUAD_TYPE is 2, this means that the ppr are exactly as indicated on the encoders.
but since the plugin can count every edge, we can achieve 4-fold resolution, which is achieved with QUAD_TYPE = 0
I don't know why you have a 5 configured.

the plugin does exactly what it is supposed to do and can also be checked in the test-bench, a QUAD_TYPE=2 divides all pulses by 4, so that in principle you only count one edge of a signal

in order to use the highest resolution cleanly, the pulses should also come cleanly and symmetrically from encoder,
~50% duty cycle at constant speed, which normal encoders also do.

what i can well imagine at the moment is that your esp32-test-encoder-tool does not output symmetrical pulses and therefore produces strange side effects

jschoch, 2025-01-01:
haha, it isn't my shitty AI addled code!

I had QUAD_ENC set to 0, but It only sent a value when count was 32.  This is why RPS would drop to 0 without the smoothing/averaging even with a pulse every 1000 microseconds.

multigcs, 2025-01-01:
ok ok, but i don't see the error here, i have add 2 additional position outputs to the quadencoder .v

position: with shift 0
position2: with shift 2
position3: without shift

[image]

```

module quadencoder
    #(
         parameter BITS = 32,
         parameter QUAD_TYPE = 0
     )
     (
         input clk,
         input a,
         input b,
         output signed [BITS-1:0] position3,
         output signed [BITS-1:0] position2,
         output signed [BITS-1:0] position
     );
    reg [2:0] a_delayed = 0;
    reg [2:0] b_delayed = 0;
    always @(posedge clk) a_delayed <= {a_delayed[1:0], a};
    always @(posedge clk) b_delayed <= {b_delayed[1:0], b};
    wire count_enable = a_delayed[1] ^ a_delayed[2] ^ b_delayed[1] ^ b_delayed[2];
    wire count_direction = a_delayed[1] ^ b_delayed[2];
    reg signed [BITS-1:0] count = 0;

    assign position = $signed(count>>>0);
    assign position2 = $signed(count>>>QUAD_TYPE);
    assign position3 = count;

    always @(posedge clk) begin
        if (count_enable) begin
            if(count_direction) begin
                count <= count + 1;
            end else begin
                count <= count - 1;
            end
        end
    end
endmodule
```

jschoch, 2025-01-01:
position is accurate, it is the interval a value get sent that is hte problem.  I will send some screenshots, recreating the halscope setup

jschoch, 2025-01-01:
This is using the bitshift and QUAD_TYPE=0,  signal emulation is 1RPS for PPR of 500.  Pulse duration should be 1ms.  Scope measures frequency of single channel 125hz

Green trace is duration, range +/- ~20 microsecond

Blue is RPS, unfiltered. 433 or zero

Red is delta counts (raw_value - last_raw_value)

My setup from yesterday is gone so I'm haveing a bit of trouble reproducing the results and i'm very confused by them.

[image]

New signals, RPS 1, PPR 500, pulse delay 5ms, scope measurs 25HZ.  Notice the delta counts is still ~26.

[image]

Last setup: RPS 1, PPR 1000, pulse delay 500us, scope measures single chan frequency of 250hz, why are counts 26 or 52?

[image]

Weirder is that yesterday everything was 32 or zero.

jschoch, 2025-01-01:
i removed the 3rd stepdir plugin, recompiled, reflashed, re-added my delta_counts pin and deltacounts seem to be behaving.  Really hard to test this.

jschoch, 2025-01-01:
pulling the timestamp header bug and retesting join following errors, maybe related to this tho?

multigcs, 2025-01-01:
puhhh, yes, maybe, i think so, the frame size was to short and cut the last bits from the encoder value.

that's why i couldn't see the error, because my configuration was such that i didn't notice the bug.

i have also modified my test setup and now have a stepper with generator to generate 1rps.

without the filter i get +-0.2 rps fluctuations

with avg8 ~ 0.04 rps

i'll try calculating the rps every 10 ms again, then the rounding errors should be less.

jschoch, 2025-01-01:
we should set lower bound on RPS, like 1RPM or .016 RPS.

Also as per the DM the alpha beta filter I am using makes really nice results.  May want to have it as a general purpose function and add to a "filter library"

closed 2025-04-09 by multigcs