multigcs/riocore #45: last joint following error

jschoch, 2024-12-31:
I seem to be getting following errors on the last joint configured, and the whole machine state gets really wierd.  Seems like some data getting corrupted somehow.  On my lathe I had to add a 3rd joint, and if I run the EMC logo test gcode it immediately following errors.  if I don't use joint_2 it seems to run ok..  very odd.

jschoch, 2025-01-01:
it is also happening on JOINT_1   here is a screenshot, note how it just shot off to the right out of nowhere.
[image]

jschoch, 2025-01-01:
and here you can see the velocity goes to negative 127 which it shouldn't be able to do based on the config
[image]

jschoch, 2025-01-01:
another where X just took off.  first time i've seen a following error on X
[image]

multigcs, 2025-01-01:
sorry, found the bug: https://github.com/multigcs/riocore/commit/859c3d1796bdcc8372b579f125af03ccc82447d1

jschoch, 2025-01-01:
pulled that commit,  maybe this is TP related or gcode post related.  going to try to remove some lines and retest
[image]

jschoch, 2025-01-01:
i can reproduce the error with this gcode

```gcode
%
(1 noface fix rub)
 G7
 G90 G18
 G21
(PROFILE ROUGHI 2)
 T1 M6
 G90 G95 G18
 G54
 G97 S1778 M3
 G0 X35.8 Z59.7
 G96 D3500 S200 M3
 G1 X31.2 F0.04
 G0 Z59.7 (Changed from: "Z59.7")
 G1 X21.
 G1 Z55.68 (Changed from: "Z55.68")
 G1 X22. Z55.56
 G0 Z59.7 (Changed from: "Z59.7")
%

seems to be related to the direction of the stepper is my guess.  going to try to scope it to see what is happening.

```

jschoch, 2025-01-01:
Here the rio.JOINT1.position jumps to ~42 in about 1ms, then you can see the purple velocity scream to -300 and ferror goes negative.

[image]

jschoch, 2025-01-01:
I rebuilt the config from scratch and i'm adding plugins back in.  will reopen if this starts happening again but for now it seems stable with no joint errors.  I have a sneaking suspicion it has something to do with wled plugin, haven't added that back in yet.

closed 2025-01-01 by jschoch