multigcs/riocore #52: Feature Suggestion: board transported UART/RS485/etc.

jediminer543, 2025-02-21:
Working on setting up a mill with a rs485 compatible vfd, at current project state this appares requires two cables from the mill to the box operating LinuxCNC/rio.

It should be possible to put a UART or RS485 block inside the FPGA and then transport the data sent to a file on the LinuxCNC side to the board and vice versa. (with the obvious addition of a RS485 tranciever IC somewhere in the loop should that be used in the future)

### Things needed for this to work

- UART core for FPGA (should already exist in the project)
- A linux side method of forwarding the data from a file on the file system to the FPGA board
- Some way to configure this in the UI

Going to be looking into the complexities of doing this myself and contributing, but wanted to open this for discussion

multigcs, 2025-02-22:
which vfd exactly is that ?

jediminer543, 2025-02-22:
I was intending on making this as a general purpose suggestion for any linuxcnc compatible kit that may be used regardless of plugin type, but in my case it is specifically a HY01D523B; this is supported in LinuxCNC by the hy_vfd component. In the current usecase I am running UDP-SPI so I would like to have the one ethernet cable to hook the machine up and be able to acquire all the relevent machine statistics from that.

I am aware I can configure the VFD to run from an external frequency source, but I would like to also be able to have configuration parameters available to LinuxCnc, which RS485 permits (and in general permits across many industrial components); this count by copied into rio as a plugin but then one would need to add dedicated logic for every RS485/UART device that is permitted by LinuxCNC, whereas bridging a system side file to the board would trivially solve this

There's already a UART bridge; and it might be possible to just do some LinuxCNC side C logic to map the nets to file but I'm not sure there's a way to do this currently, and might need to do linux side buffering.

multigcs, 2025-02-22:
feature request is rejected, ................ already exists ;)

https://github.com/multigcs/riocore/tree/main/riocore/plugins/modbus

you need 3 free pins and something like this:

https://www.roboter-bausatz.de/p/max485-modul-rs-485-ttl-auf-rs485-modul

jediminer543, 2025-02-22:
Oh that's neat, I'd not seen that, thanks :3

This still has the general implementation issue of needing arb. support but solves my issue so yeah, thankies

closed 2025-02-22 by jediminer543