multigcs/riocore #53: Adding new board

goran-mahovlic, 2025-04-08:
Hi, 

thank you on this great project!
I would like to add more ECP5 boards like ULX3S, and maybe later on ULX4M boards, could you please point me how to add board, and how to test it?
I also have OrangeCrab at home so I can also add/test...

Goran

multigcs, 2025-04-08:
Hi,
thanks :)
i have add a short doc:
https://github.com/multigcs/riocore/blob/dev/doc/BOARDS.md

If you have any further questions, just ask !

goran-mahovlic, 2025-04-08:
> Hi, thanks :) i have add a short doc: https://github.com/multigcs/riocore/blob/dev/doc/BOARDS.md
> 
> If you have any further questions, just ask !

Perfect! Thank you! I did not expect so fast response, will try to add ULX3S next week...

There is good progress with opensource toolchain for GateMate chip and I also have some add-ons for Olimex GateMate and I am now working on modular Gatemate board so after ULX3S maybe we could even try to add Olimex Gatemate...

https://github.com/intergalaktik/Extension_Boards_for_Olimex_GateMate

I can also easily make some extensions for that board if needed...

multigcs, 2025-04-09:
another option would be espnow:

https://github.com/espressif/esp-now
https://randomnerdtutorials.com/esp-now-esp32-arduino-ide/
https://github.com/thomasfla/Linux-ESPNOW

runs very stable on my pendant, but the main problem is the linux side,
you need to go over serial to the 'master-esp' and i have not yet got Linux-ESPNOW to run properly.

So it's not a good option for Reatime and you have the same problem with BLE.

emard, 2025-04-09:
Is there some small example of protocol traffic
over SPI what typical data goes to/from SPI and
timestamps of that so we can see if we can
match this by some available protocol

multigcs, 2025-04-09:
the packages depends on the configuration (riocore/configs/Tangbob/config.json):
 https://cserver1.multixmedia.org/interface-example.html

interval is 1ms and always same size

multigcs, 2025-04-09:
Here is an UDP/SPI bridge that we used before:
https://github.com/multigcs/LinuxCNC-RIO/blob/main/UDP2SPI-Bridge/WT32-ETH01/src/main.ino

emard, 2025-04-09:
I see, 1 ms is difficult to match with BLE, also WiFi is
unreliable on long term. If the 1 ms is configurable so one
can choose something 1-50 ms for slow jobs then it would
be maybe even possible wireless

I think cool option is to use SPDIF optical cable,
because cables and trasncievers are cheap and
optics makes it electrically well decoupled from high
noise motors

multigcs, 2025-04-09:
> I think cool option is to use SPDIF optical cable, because cables and trasncievers are cheap and optics makes it electrically well decoupled from high noise motors

cool idea

>I see, 1 ms is difficult to match with BLE, also WiFi is unreliable on long term. If the 1 ms is configurable so one can choose  something 1-50 ms for slow jobs then it would be maybe even possible wireless

it is configurable but not recommended to go higher, but feasible for testing or for applications other than linuxcnc

multigcs, 2025-04-09:
but, can i send custom frames to SPDIF in linux ?
SPI2SPDIF bridge could be nice

emard, 2025-04-09:
SPDIF can do 48kHz 24-bit mono or stereo
PCM playback with peak bandwidth of 0.768 Mbps

SPDIF PCM Protocol is very "light" and open (compared to
DTS which closed). So FPGA can easily extract
binary data from PCM playback stream. What you binary
play in PCM (wav) stream exactly comes on wire
od optical cable and FPGA gets back same binary

Linux and other OS have good realtime support for sound
playback so SPDIF PCM2SPI bridge on FPGA should
work great for CNC.

emard, 2025-04-10:
https://www.tinysineaudio.com/products/toslinkbee-spdif-to-i2s-digital-audio-receiver-module
interesting is this, SPDIF to I2S receiver module
I2S is similar to SPI

jschoch, 2025-04-17:
if you get a s3 you can do ethernet/udp or tcp via tinyusb.  It isn't crazy fast but it is super easy to setup.  my crappy pendant does esp-now -> S3 esp-now -> tinyusb/udp and then i have a linuxcnc comp that runs the UDP server.